simonsobs/ocs #59: Prevent Operations from exiting with "job" mutex set

mhasself, 2019-05-24:
The boiler-plate for Agents shows to use try_set_job to mark the active job, and then each Operation code will clear the .job on exit.  The implementation arose in examples where a simple mutex (no two Operations could run simultaneously) was desired, and errors were unlikely...  But for real agents, we want a system that clears the mutex when the Operation exits, whether it was clean or not.  Probably we should have some decorators that implement a try/finally, so the state never gets stuck.  The decorator could also handle the basic function of rejecting start requests for Operations that are already in progress.

The issue of inter-compatibility of Operations (what things are allowed or forbidden or required to run simultaneously) needs to be dealt with carefully -- get a lock, inspect the list of currently running agents and decide whether it's safe to start the present Operation, release the lock, proceed or exit with error.

jlashner, 2019-05-24:
My solution to this was the [TimeoutLock](https://github.com/simonsobs/ocs/blob/98d692137a0fc4c9f394685ad303bdf68e25aa03/ocs/ocs_twisted.py#L4), which would always release on exit, even if there's an error. (Here's)[https://github.com/simonsobs/socs/blob/c49b13bf4ea979f56edb374ced70e3f9b5e6951c/agents/keithley2230G-psu/keithley_agent.py#L40] an example of how its used in the keithley psu agent. Calling acquire_timeout(0) acts the same way as try_set_job did, where it will return with False immediately if the lock is already in use. Should we switch to using this in the example agents?

mhasself, 2019-07-24:
Closing, as TimeoutLock seems to have broadly fixed the problem.

closed 2019-07-24 by mhasself