simonsobs/ocs #60: Aggregator prov_id list sync issue

BrianJKoopman, 2019-06-21:
We've seen a similar bug to #36 where the aggregator loses track of a provider under certain circumstances and fills the logs with errors like:

```
2019-06-21T20:16:19+0000 OCSAgent.onUserError(): “While firing <bound method DataAggregator._data_handler of <__main__.DataAggregator object at 0x7fba72af7198>> subscribed under 6560288436609885.”
Traceback (most recent call last):
 File “/usr/local/lib/python3.6/dist-packages/autobahn/twisted/websocket.py”, line 322, in _onMessage
   self.onMessage(payload, isBinary)
 File “/usr/local/lib/python3.6/dist-packages/autobahn/wamp/websocket.py”, line 95, in onMessage
   self._session.onMessage(msg)
 File “/usr/local/lib/python3.6/dist-packages/autobahn/wamp/protocol.py”, line 726, in onMessage
   future = txaio.as_future(handler.fn, *invoke_args, **invoke_kwargs)
 File “/usr/local/lib/python3.6/dist-packages/txaio/tx.py”, line 429, in as_future
   return maybeDeferred(fun, *args, **kwargs)
--- <exception caught here> ---
 File “/usr/local/lib/python3.6/dist-packages/twisted/internet/defer.py”, line 151, in maybeDeferred
   result = f(*args, **kw)
 File “aggregator_agent.py”, line 168, in _data_handler
   prov_id = self.prov_ids[feed[“address”]]
builtins.KeyError: ‘observatory.LSA24MA.feeds.temperatures’
```

I don't know how to reproduce this yet, but typically it occurs when someone goes wrong with the Agent like in [socs Issue #6](https://github.com/simonsobs/socs/issues/6). The LSA24MA 240 Agent was restarted, but the aggregator still produced these errors.

Aggregator version: v0.5.0-29-g2265349

BrianJKoopman, 2019-07-02:
Alright, so we've figured out how to reproduce this bug. Start with a freshly started aggregator and 240 Agent. Then restart the 240 Agent via a `docker-compose restart <agent name>` command. What occurs is when the 240 Agent is stopped, the aggregator schedules the removal of the associated provider. Then the 240 Agent is restarted, the aggregator realizes the `prov_id` is different and replaces it. Then the scheduled removal occurs, removing the new instance of the provider with updated `prov_id`. Here's more complete log output:

```
2019-07-02T18:45:46+0000 Creating file: /data/15620/2019-07-02-18-45-46.g3                                                                                                                                                                                                                                                    
2019-07-02T18:46:44+0000 Scheduled remove for provider observatory.LSA24MA.feeds.temperatures                                                                                                                                                                                                                                 
2019-07-02T18:46:45+0000 Somehow there is a new instance of observatory.LSA24MA.feeds.temperatures.Scheduling removal of old instance                                                                                                                                                                                         
2019-07-02T18:46:45+0000 Added provider observatory.LSA24MA.feeds.temperatures (agent sess: 1562093204.9089541) with id 4                                                                                                                                                                                                     
2019-07-02T18:46:45+0000 Feed observatory.LSA24MA.feeds.temperatures is already subscribed to                                                                                                                                                                                                                                 
2019-07-02T18:46:45+0000 Removing provider observatory.LSA24MA.feeds.temperatures with agent id 1562019610.397801                                                                                                                                                                                                             
2019-07-02T18:49:05+0000 OCSAgent.onUserError(): "While firing <bound method DataAggregator._data_handler of <__main__.DataAggregator object at 0x7f8af39e0198>> subscribed under 1145008383491662."                                                                                                                          
Traceback (most recent call last):                                                                                                                                                                                                                                                                                            
  File "/usr/local/lib/python3.6/dist-packages/autobahn/twisted/websocket.py", line 322, in _onMessage                                                                                                                                                                                                                        
    self.onMessage(payload, isBinary)                                                                                                                                                                                                                                                                                         
  File "/usr/local/lib/python3.6/dist-packages/autobahn/wamp/websocket.py", line 95, in onMessage                                                                                                                                                                                                                             
    self._session.onMessage(msg)                                                                                                                                                                                                                                                                                              
  File "/usr/local/lib/python3.6/dist-packages/autobahn/wamp/protocol.py", line 726, in onMessage                                                                                                                                                                                                                             
    future = txaio.as_future(handler.fn, *invoke_args, **invoke_kwargs)                                                                                                                                                                                                                                                       
  File "/usr/local/lib/python3.6/dist-packages/txaio/tx.py", line 429, in as_future                                                                                                                                                                                                                                           
[... cut ...]      
builtins.KeyError: 'observatory.LSA24MA.feeds.temperatures'                                                                                                                                                  
```

A simple solution that occurs to me is the scheduled removal should check the `prov_id`. If it is already, then something else is going on.

This should be a high priority to fix. Agents can be restarted reliably if done slowly, but this prevents the use of `docker-compose restart`.

jlashner, 2019-07-02:
Just looked over the code and I think I know what's wrong. I'll try to fix it today, thanks for figuring out how to recreate it.

BrianJKoopman, 2019-07-03:
Great, thanks for the patch. I just tested it with one of our 240s and it seems to work.

My one comment (which is necessarily related to the bug, but while we're here fixing related thing...) is that we should use a uniform name for 'agent_session_id', the logs refer to 'agent sess' and 'agent id'. It seems both are an abbreviation of 'agention_session_id', but we should be consistent, otherwise the casual observer might not know they're actually the same thing.

Please address that, then merge away.

closed 2019-07-24 by jlashner