simonsobs/ocs #75: Agents are unaware of their registry status

BrianJKoopman, 2019-09-30:
## Issue
Agents are currently unaware of their registration status. If they become unregistered for some reason while they are still functional they do not know and thus do not attempt to reestablish their registration with the registry agent. This will currently cause any data they publish to not be saved the disk by the aggregator.

## Proposal
Agents should subscribe to the "agent_activity" feed and watch for themselves to be unregistered. If they are, for whatever reason, unregistered they should try to reestablish their registration with the registry agent.

## Related Issues
This is somewhat related to #67. The agents should be able to re-register themselves if the registry restarts. If the registry doesn't broadcast removals before restart then probably this will require a separate solution.

#74 is related in that the state that is caused by these two issues was observed by @Zhilei during testing.

mhasself, 2019-09-30:
If I may counter-propose, I think the feed subscription scheme might already be over-complicated, and we should take this opportunity to make a simpler, more robust system (rather than adding a layer to handle drops+reconnects).

Can we explore the possibility of discarding registration and heartbeats... and instead having Agents announce their complete registration information regularly, on a timer?  This will provide both a heartbeat and whatever information the aggregator / registry need to initialize agent or feed handling.  If anything dies, it need only listen for a moment to learn about everything that is running on the system.

BrianJKoopman, 2019-10-01:
Definitely! I wanted to open the issue for discussion.

I do agree that maybe things are a bit over-complicated right now. I think our goal in introducing the registry was to have a central authority for what agents were running. Currently the way an agent tells the registry about itself is by calling one of the registry agent's tasks. In doing so it describes itself with this info:
```
        return {
            'agent_address': self.agent_address,
            'session_id': self.agent_session_id,
            'feeds': [f[1].encoded() for f in self.feeds.items()],
            'tasks': list(self.tasks.keys()),
            'processes': list(self.processes.keys())
        }   
```

So we could publish this as the replacement heartbeat every X seconds and remove the registry altogether. We'd still have to somehow tell the aggregator what URI to subscribe to in order to listen for this info.

jlashner, 2019-10-01:
I'm not really here, but you can subscribe to feeds with wildcards, so anything with the feedname  `root.<instance>.heartbeat`

mhasself, 2019-10-02:
Currently we have to tell all agents the address of the registry (so they can call the registration tasks).  Perhaps we could retain that setting, but have agents all yell their status on some single feed in that namespace (``observatory.registry.feeds.agent_announce``?).

Some mods for the announcement info:
- add "announcement_expires", timestamped, which a watcher can use to timeout (instead of hard-coding "1 second" or "5 seconds" all over the place.
- add a string for the Agent class (i.e. LS240Agent)... what do we usually call this?  agent_type?  agent_class?

The registry Agent should not be discontinued, yet, since it's already part of everyone's system... rather, let's migrate it to be more of a monitor, keeping an eye on the agent announcements in order to provide concise summaries of observatory status.

mhasself, 2019-11-06:
The HK aggregation problem that led to this issue is resolved in #78 .  The role of the registry, going forward, is still up in the air.  I guess we should clarify that before closing this issue (either in a PR or another issue!).

jlashner, 2020-03-10:
This is solved by the registry update in PR #109. With the update, agents no longer need to call a `register_agent` task. Instead, the registry monitors all heartbeats and automatically registers/expires agents. If the registry restarts, it will automatically pick up all agents on the network as soon as it here's their heartbeat.

closed 2020-03-10 by jlashner