simonsobs/ocs #92: Lots of required packages

jlashner, 2019-11-08:
`pip install -r requirements.txt` Currently installs a lot of packages
```

BrianJKoopman, 2019-11-08:
Just taking a quick look there are a few old packages in the requirements file, such as the `pyserial` for the 240's. Everything in the top group is used (I can find imports for them in the current master branch) except for `msgpack`, what was that used for?

I think it's common to see developer packages split off, so we could move things under `# testing` to a separate file, that'll help a little bit.

Have you identified the set required for running clients? If I had to guess it'd be almost everything in the top group except for maybe crossbar. That does seem to be the biggest culprit for pulling in dependencies though, installing in a fresh virtualenv:
```
Installing collected packages: idna, setproctitle, pygments, click, pyasn1, pycparser, cffi, cryptography, pyasn1-modules, service-identity, cbor2, numpy, txaio, pynacl, pyyaml, lmdb, zlmdb, psutil, u-msgpack-python, pyopenssl, sortedcontainers, pytrie, priority, pyqrcode, werkzeug, hyperframe, hpack, h2, lz4, python-snappy, argon2-cffi, passlib, cbor, ujson, flatbuffers, py-ubjson, msgpack, autobahn, txtorcon, bcrypt, sdnotify, argh, pathtools, watchdog, bitstring, netaddr, MarkupSafe, jinja2, mistune, certifi, chardet, urllib3, requests, treq, crossbar
  Found existing installation: idna 2.8
    Uninstalling idna-2.8:
      Successfully uninstalled idna-2.8
Successfully installed MarkupSafe-1.1.1 argh-0.26.2 argon2-cffi-19.2.0 autobahn-19.11.1 bcrypt-3.1.7 bitstring-3.1.6 cbor-1.0.0 cbor2-4.1.2 certifi-2019.9.11 cffi-1.13.2 chardet-3.0.4 click-7.0 crossbar-19.10.1 cryptography-2.8 flatbuffers-1.11 h2-3.1.1 hpack-3.0.0 hyperframe-5.2.0 idna-2.5 jinja2-2.10.3 lmdb-0.98 lz4-2.2.1 mistune-0.8.4 msgpack-0.6.2 netaddr-0.7.19 numpy-1.15.4 passlib-1.7.1 pathtools-0.1.2 priority-1.3.0 psutil-5.6.5 py-ubjson-0.14.0 pyasn1-0.4.7 pyasn1-modules-0.2.7 pycparser-2.19 pygments-2.4.2 pynacl-1.3.0 pyopenssl-19.0.0 pyqrcode-1.2.1 python-snappy-0.5.4 pytrie-0.3.1 pyyaml-5.1.2 requests-2.22.0 sdnotify-0.3.2 service-identity-18.1.0 setproctitle-1.1.10 sortedcontainers-2.1.0 treq-18.6.0 txaio-18.8.1 txtorcon-19.1.0 u-msgpack-python-2.5.2 ujson-1.35 urllib3-1.24.3 watchdog-0.9.0 werkzeug-0.16.0 zlmdb-19.9.3
```

BrianJKoopman, 2021-12-02:
We dropped crossbar a while back, which did reduce many of the dependencies pulled in. More details on why this should be considered resolved in [#167.](https://github.com/simonsobs/ocs/pull/167#issuecomment-984827869)

closed 2021-12-02 by BrianJKoopman