simonsobs/ocs #102: Support running some Tasks during data acq Processes

mhasself, 2020-01-21:
The following situation occurs in a number of agents (e.g. Power Supplies, Lakeshore readout):

- There is a Process for data acquisition.
- There are many Tasks for manipulating device settings.
- One can't call those Tasks without suspending data acquisition.

This is annoying because even when settings Tasks don't change the readout characteristics, the acq process has to be stopped and started.

A general solution is have the acq Process not hold the main job lock.  Instead, it would acquire (and occasionally release and reacquire) a special "sampling_lock".  Any Tasks that want to update device settings must first acquire that sampling_lock (which would then ensure no conflict with the acq Process).

Not sure to what extent this requires some library support vs. just some hacking on the agents in question.

jlashner, 2020-01-22:
Should we worry about releasing and reacquiring the sampling lock @ 10Hz? Maybe we should release it every 10 data points or something if there is a high-ish data rate.

jlashner, 2020-01-22:
Also why does it have to be a separate lock? Can it be the same as the main job lock?

mhasself, 2020-01-22:
I guess in simple cases it can be the single main job lock!  And perhaps even in slightly more complex cases.  I see now what you have done in the Keithley agent, that seems totally fine.

Yes, let's baseline having the acq process release the lock every 1 second or so, but not more often than that.

mhasself, 2020-05-13:
Agents needing this functionality may find the new form of TimeoutLock useful -- see #125.  The Lakeshore372 agent is a working application.

closed 2020-05-13 by mhasself