simonsobs/ocs #124: Agent access control mechanism

mhasself, 2020-04-15:
As discussed on call today, it would be good to have a way of implementing privileges, mostly to prevent accidental manipulation of an Agent when coordinated events are happening (such as operation of the Observatory by the main Sequencer).

It's not super obvious how to do this.  One way would be to have all clients send an access level (as an integer, say) with all Operation method calls.  This would happen as part of the underlying protocol and not be accessible as an argument in the call signature.  On the receiving end, the access level would be checked by OCSAgent code (not in the Agent's Operation implementation) and rejected if the access level is insufficient for the Agent's current minimum access level.

Finally... the Agent's minimum access level would be determined by it monitoring some pubsub channel(s).  When a Sequencer starts working, it publishes messages of the form "hey all agents, raise your minimum access level to 5 for the next 2 minutes or until I let you know otherwise".  When the Sequencer wants to relinquish exclusive access, it publishes "hey all agents, you may lower your minimum access level to 0."

We're mostly trying to avoid _accidental conflicts_ here -- a given MatchedClient should default to access level 0 but should be able to escalate to level 5, say, through a keyword argument or data member set.

Does this make sense?  Discuss.

newburgh, 2020-04-16:
FYI Your original discussion of this issue was during the review in April 2018, your slides are here with ~3 pages regarding exclusive access:
https://drive.google.com/file/d/1EPe9QPCwHKO1TD7odXNY8952Qr19ehPS/view?usp=sharing

mhasself, 2020-04-16:
Thanks for the reminder on that!  Of the two suggestions in that talk, I don't really like Approach 1.  I think Approach 2 (allowing users to "check out" subsystems) is quite compatible with the "Access Level" ideas.

closed 2026-02-27 by BrianJKoopman