simonsobs/ocs #130: Validate field names in hk_aggregator

mhasself, 2020-05-12:
There are rules for field names defined in the tod_format doc.  We should consider revising them, and then check for validity in the aggregator so invalid fields don't end up in the data.

jlashner, 2020-05-13:
What rules are you thinking of? I think we should definitely stop using spaces if field names and check against that, but is there anything else in particular?

mhasself, 2020-05-13:
Current rules are very strict:  _Names of fields and blocks should follow certain rules to be valid. A valid name:_
_– contains only letters (a-z, A-Z; case sensitive), decimal digits (0-9), and the
underscore (\_).
– begins with a letter, or with any number of underscores followed by a letter.
– is no more than 255 characters long._

This is strict... but based on:
- The "." is forbidden because it's a domain separator in OCS agent and feed names.
- One might want to parse arithmetic expressions with field names, so it's perhaps wise to outlaw +/-/*//.
- For easier string parsing of lists of field names, we would be wise to outlaw whitespace and "," and ";".

Ultimately is easier to define by inclusion rather than exclusion, which led to the above rules, which allow for _ as the only separator.

closed 2020-06-18 by BrianJKoopman