simonsobs/ocs #174: reparse_args causes Agents that use it to crash

BrianJKoopman, 2020-09-30:
The recent addition of the [class_name attribute](https://github.com/simonsobs/ocs/blob/5c3325a4c3afed52b9c07d315dba4319a570b15c/ocs/ocs_agent.py#L40) in the base OCSAgent causes older Agents still using the deprecated [site_config.reparse_args](https://github.com/simonsobs/ocs/blob/5c3325a4c3afed52b9c07d315dba4319a570b15c/ocs/site_config.py#L520) method to crash on startup.

```
2020-09-30T14:39:22+0000 Traceback (most recent call last):
2020-09-30T14:39:22+0000   File "influxdb_publisher.py", line 291, in <module>
2020-09-30T14:39:22+0000     agent, runner = ocs_agent.init_site_agent(args)
2020-09-30T14:39:22+0000   File "/app/ocs/ocs/ocs_agent.py", line 42, in init_site_agent
2020-09-30T14:39:22+0000     class_name=getattr(args, 'agent_class'))
2020-09-30T14:39:22+0000 AttributeError: 'Namespace' object has no attribute 'agent_class'
```

Since this has been marked as deprecated for almost a year now, it's good motivation to remove all uses of `reparse_args` from Agents.

mhasself, 2020-09-30:
Sorry about that... the whole point of my using getattr was for backwards compatibility... I meant to put ```getattr(args, 'agent_class', None)```.

If backwards compatibility is now out the window anyway, then that argument in ocs_agent.py should be changed to ```class_name=args.agent_class```.

BrianJKoopman, 2020-10-01:
Thanks for the patch. I merged it in #177, though we should still work on removing the deprecated `reparse_args`.

PR #177 (2020-10-01): Add default agent_class attribute to agent instatiation

## Description

Very small change related to #174.

## Motivation and Context

fixes #174

## How Has This Been Tested?

Tested in isolated set of containers, one of which was the influxdb publisher which broke from #174 and is fixed by this change.

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Checklist:

- [ ] My code follows the code [... cut]
commit 488bc7cf (2020-10-01): Add default agent_class attribute (#177)

fixes #174
closed 2020-10-01 by BrianJKoopman