simonsobs/ocs #203: site_configs.parse_args isn't using site-config-file values

BrianJKoopman, 2021-05-05:
While making some changes to an socs Agent related to parsing a value from the SCF, I'm finding that the newer `site_configs.parse_args` isn't taking the SCF value.

The argument is `--auto-acquire`, which is being set to `False` in the SCF.

In the Agent previously was:
```
    # Parse arguments
    site_parser = site_config.add_arguments()
    parser = make_parser(site_parser)
    args = parser.parse_args()
    site_config.reparse_args(args, 'Lakeshore370Agent')

    agent, runner = ocs_agent.init_site_agent(args)

    print(args.auto_acquire)
    # Configure automatically acquiring data
    init_params = {'auto_acquire': args.auto_acquire}
```

`init_params` is then passed as the `startup` argument when registering the `init_lakeshore` task. This behaves as expected, `args.auto_acquire` is `False`, and the acquisition process doesn't start:
```
2021-05-05T21:55:06+0000 LS370_agent.py:761: deprecation.DeprecatedWarning: reparse_args is deprecated as of v0.6.0. Use site_config.parse_args instead
2021-05-05T21:55:06+0000 site_config is setting values of "serial_number" to "370A7E".
2021-05-05T21:55:06+0000 site_config is setting values of "port" to "/dev/ttyUSBA7E".
2021-05-05T21:55:06+0000 site_config is setting values of "auto_acquire" to "False".
2021-05-05T21:55:06+0000 Using OCS version 0.7.1+17.g1162576
2021-05-05T21:55:06+0000 ocs: starting <class 'ocs.ocs_agent.OCSAgent'> @ observatory.370A7E
2021-05-05T21:55:06+0000 log_file is apparently None
2021-05-05T21:55:06+0000 False
2021-05-05T21:55:06+0000 transport connected
2021-05-05T21:55:06+0000 session joined: 
SessionDetails(realm="test_realm",
               session=4700894831771937,
               authid="6M5X-TNAJ-6YEQ-F3CX-VSHY-P3HU",
               authrole="iocs_agent",
               authmethod="anonymous",
               authprovider="static",
               authextra={'x_cb_node': '47ccea84caf4-1', 'x_cb_worker': 'worker001', 'x_cb_peer': 'tcp4:10.0.1.26:37076', 'x_cb_pid': 16},
               serializer="cbor.batched",
               transport="websocket",
               resumed=None,
               resumable=None,
               resume_token=None)
2021-05-05T21:55:06+0000 startup-op: launching init_lakeshore
2021-05-05T21:55:06+0000 start called for init_lakeshore
2021-05-05T21:55:06+0000 init_lakeshore:0 Status is now "starting".
2021-05-05T21:55:06+0000 init_lakeshore:0 Status is now "running".
2021-05-05T21:55:06+0000 9600
2021-05-05T21:55:12+0000 Initialized Lakeshore module: <socs.Lakeshore.Lakeshore370.LS370 object at 0x7fcef0ec0f70>
2021-05-05T21:55:12+0000 init_lakeshore:0 Lakeshore initilized with ID: LSCI,MODEL370,370A7E,04102008
2021-05-05T21:55:12+0000 init_lakeshore:0 Lakeshore module initialized.
```

Updating the Agent to try to use `site_config.parse_args` it becomes:
```
    # Parse arguments
    parser = make_parser()
    args = site_config.parse_args(agent_class='Lakeshore370Agent', parser=parser)

    agent, runner = ocs_agent.init_site_agent(args)

    print(args.auto_acquire)
    # Configure automatically acquiring data
    init_params = {'auto_acquire': args.auto_acquire}
```

This results in:
```
2021-05-05T21:55:46+0000 Using OCS version 0.7.1+17.g1162576
2021-05-05T21:55:46+0000 ocs: starting <class 'ocs.ocs_agent.OCSAgent'> @ observatory.370A7E
2021-05-05T21:55:46+0000 log_file is apparently None
2021-05-05T21:55:46+0000 True
2021-05-05T21:55:46+0000 transport connected
2021-05-05T21:55:46+0000 session joined: 
SessionDetails(realm="test_realm",
               session=5534896136516717,
               authid="Y6PQ-PUSH-SVGV-S3WC-PRSX-3PH9",
               authrole="iocs_agent",
               authmethod="anonymous",
               authprovider="static",
               authextra={'x_cb_node': '47ccea84caf4-1', 'x_cb_worker': 'worker001', 'x_cb_peer': 'tcp4:10.0.1.27:50682', 'x_cb_pid': 16},
               serializer="cbor.batched",
               transport="websocket",
               resumed=None,
               resumable=None,
               resume_token=None)
2021-05-05T21:55:46+0000 startup-op: launching init_lakeshore
2021-05-05T21:55:46+0000 start called for init_lakeshore
2021-05-05T21:55:46+0000 init_lakeshore:0 Status is now "starting".
2021-05-05T21:55:46+0000 init_lakeshore:0 Status is now "running".
2021-05-05T21:55:46+0000 9600
2021-05-05T21:55:51+0000 Initialized Lakeshore module: <socs.Lakeshore.Lakeshore370.LS370 object at 0x7f6f01473310>
2021-05-05T21:55:51+0000 start called for acq
```
`args.auto_acquire` remains `True`, and the acquisition task is started, despite `--auto-acquire` being `False` in the SCF.

jlashner, 2021-05-05:
what branch are you in / can you post the site-config entry?

jlashner, 2021-05-05:
My first thought is that this is a problem with how auto-acquire is being parsed, since passing bools through argparse is kind of weird and you need to use the `action="store_true"`  option

jlashner, 2021-05-05:
Ya looks like that's it, if you setup the argument like this:
```
    pgroup.add_argument('--auto-acquire', type=bool, default=True,
                        help='Automatically start data acquisition on startup')
```
ArgParse will actually parse `False` as the string `"False"` which evaluates to `True` when you convert to a boolean. lol

BrianJKoopman, 2021-05-05:
> My first thought is that this is a problem with how auto-acquire is being parsed, since passing bools through argparse is kind of weird and you need to use the `action="store_true"` option

Ah, yeah, that makes sense. I'll switch to `store_true`, should have tried that first...Thanks for the quick response.

closed 2021-05-05 by BrianJKoopman