simonsobs/ocs #215: Help Operations to extract params and propagate errors

mhasself, 2021-08-11:
(As discussed [here](https://github.com/simonsobs/socs/pull/178#discussion_r685253100).)

It is agreed that it would be helpful if ocs library could help Agent Operation functions to process params.  This can be accomplished with some helper function(s) to extract parameters from the params dict, along with some exception classes for the various problems that might arise (maybe just ParamInvalid is enough).  In cases where the helper function(s) raise ParamInvalid, it will get caught in OCSAgent, which will mark the Operation as failed and propagate a useful message back to the user.

mhasself, 2021-08-20:
I have two approaches developed for this ... here's how they'd look like in the FakeDataAgent:

1. Instantiate handler and extract parameters with instructions at the top of the operation function:
```
    @inlineCallbacks
    def delay_task(self, session, params={}):
        params = ocs_agent.ParamHandler(params)                                                        
        delay = params.get('delay', 5., type=float, check=lambda x: 0 < x < 100)                       
	succeed = params.get('succeed', default=True) is True                                          
        params.check_for_strays()                                                                      
        ...
```
2. Decorate the agent function, and then be free to assume all parameters are properly checked and processed:
```
    @ocs_agent.param('delay', 5., type=float, check=lambda x: 0 < x < 100)
    @ocs_agent.param('succeed', default=True)
    @inlineCallbacks
    def delay_task(self, session, params={}):
        delay = params['delay']  # guaranteed to be a float between 0 and 100
        succeeded = params['succeed'] is True  # default will have been subbed in
```

The method (2) is a little uglier to implement, but it has one really important advantage: *the parameters can be checked immediately when the client calls `start()`*, and return an ERROR in response to the start() if the params are not compliant.

I.e. in method (2) you would get this:

```
  >>> c.delay_task.start(delay='invalid')
  OCSReply: ERROR : Param 'delay'=invalid is not of required type (<class 'float'>)
     (no session -- op has never run)
```
whereas with method (1) you'd get:
```
  >>> c.delay_task.start(delay='invalid')
  OCSReply: OK : Started task "delay_task".
    delay_task[session=2]; status=starting for 0.009418 s
    messages (1 of 1):
      1629477150.242 Status is now "starting".
    other keys in .session: op_code, data
  >>> c.delay_task.wait(timeout=1.)
  OCSReply: OK : Operation "delay_task" is currently not running (FAILED).
    delay_task[session=0]; status=done with ERROR 0.113139 s ago, took 0.000708 s
    messages (3 of 3):
      1629471108.736 Status is now "starting".
      1629471108.737 ERROR: Param 'delay'=invalid is not of required type (<class 'float'>)
      1629471108.737 Status is now "done".
  other keys in .session: op_code, data
```

@BrianJKoopman @jlashner  I'd be interested in your thoughts on this.  We can make both of these solutions available but we should probably encourage one or the other.  I think you like decorators more than I do so maybe there's no hesitation for method (2).

The potential drawback of (2) is if you have more complicated param validation, such as when the permissible values of one parameter depend on the value of some other parameter.  Those cases would tend to break simple validation anyway... but the point is, now a user would have to be wary of parameter troubles arising on start() and on wait().)

jlashner, 2021-08-20:
Yeah I agree that failing on the client-side as soon as start is called is a huge benefit, and I definitely think that we should recommend that people use method 2. I think decorators are fine, though combining decorators + lambda functions seems like it can get ugly pretty fast. If you don't pass in a param that doesn't have a default value will that also raise an error?

BrianJKoopman, 2021-08-20:
I do like (2) better. Having the check happen immediately is nice, and (2) also looks cleaner to me. Though I suppose if there are a dozen params then the decorator list is going to get kind of long...

> The potential drawback of (2) is if you have more complicated param validation, such as when the permissible values of one parameter depend on the value of some other parameter. Those cases would [...]

Are you saying this breaks in (1) too anyway? If that's the case I'd rather just only support (2). Is there good reason to support both? What I anticipate happening if we support both is one method will be used first, then copied into all future Agents and essentially be the only one used.

mhasself, 2021-08-21:
> If you don't pass in a param that doesn't have a default value will that also raise an error?

Ya, any param for which you don't set a default is enforced as mandatory.  Also, by default, passing something=None is the same as not passing something at all (that can be overridden but I think it's a sensible behavior even if it's stricter than Python would be).  But you can set default=None.

> Are you saying this breaks in (1) too anyway? If that's the case I'd rather just only support (2). Is there good reason to support both? What I anticipate happening if we support both is one method will be used first, then copied into all future Agents and essentially be the only one used.

I meant to emphasize the point that "complex" parameter checking, by my definition of complex, can't be done before actually calling the agent-supplied function.  In case (1) a complex param check failure would look like a simple param check failure... they'd both show up in the session log, and be apparent when the user calls wait().  In case (2) a simple failure shows up immediately on start() but a complex failure shows up when the user calls wait().

Thanks for the feedback ... I appreciate the enthusiasm for the decorators.  I'll focus on that part.

closed 2021-09-15 by BrianJKoopman