simonsobs/ocs #238: Make Quickstart more of a "Minimal Working System" (or add an example of that?)

kmharrington, 2021-11-02:
The quickstart was great for getting an example of how stuff looks when it's running (ie. showing off grafana with fake data) but it has no saving / persistence across restarts.

Going that one more step to "just follow these instructions to get a minimal working / saving system" would be very nice and helpful. I really wanted to just get someone setup putting together an agent/client pair outside the LATRt system and the jump of "now go figure out how to make everything save" is a lot. All the systems I see to copy from have many stacked docker-compose files + many hosts so are a much more complicated reach.

arahlin, 2022-07-26:
I'll second this issue.  I put together an OCS system in the SPT lab following the example initially, and then became very frustrated when none of the changes I made in my setup were saved.  I think a clearer worked example would include all the docker settings that are needed for persistent storage, because it's not immediately clear (especially for those that don't use docker regularly) that all your work is destroyed when the docker is brought down.  I actually ended up dropping docker from the setup altogether on our Ubuntu 22.04 machine, used `apt install` and `pip install` for everything except `spt3g_software` (I'm working on fixing that too ;) ), and was able to run all of the necessary services basically out of the box with `systemd`.  In the end, it's pretty clear how all the pieces talk to each other, but getting to that point was a little more confusing that I think it could be.

It also wasn't immediately obvious that I needed to run a host-manager agent with aggregator/influxdb/registry agents in order to get data into all the right places.  I think the approach of running a host manager on the main OCS machine with all of the necessary data agents is relatively straightforward, and also makes it really easy to see how to run more agents on other machines elsewhere.

I'd be happy to provide more feedback, or even help update some of the documentation if that would be useful.

BrianJKoopman, 2022-07-27:
I've tried to address this issue, and many of the issues @kmharrington pointed out back in November (list in #276.) I ran through the steps in a clean VM, but if either of you would like to check out in the [new changes](https://ocs--276.org.readthedocs.build/en/276/user/quickstart.html) please let me know what you think.

@arahlin and I discussed the other day also having a section on installation without needing Docker. I'm going to create a new issue for that, in case other users would also like to avoid Docker (though we'll still recommend its use.)

PR #276 (2022-08-08): docs: Address quickstart guide issues

## Description

This fixes several user reported issues in the quickstart guide, the big one being persistent storage of grafana/influxdb data. Other items on my list were:

- Add `sudo` to user creation steps.
- Add `-E` to sudo usage when running `docker-compose up -d`.
- Fix link to Docker installation docs.
- Remove mention of `setup.py` during installation.
- Point out Agent Reference pages when discussing the site config file.
- Persist influxdb and grafana storage in quickstart example. (Also added to the longer example under "System Configuration" [... cut]