simonsobs/ocs #295: Wrong host_master script path in v0.10.0

cnweaver, 2022-11-01:
I needed to re-install OCS recently, and updated to the latest release, v0.10.0 in the process. I was rather confused to get the following output when I attempted to start the system:

	$ ocs-local-support start
	
	Trying to launch hostmanager agent...
	Traceback (most recent call last):
	  File "/home/cweaver/ocs/bin/ocs-local-support", line 8, in <module>
	    sys.exit(main_local())
	  File "/home/cweaver/ocs/lib/python3.8/site-packages/ocs/ocsbow.py", line 954, in main_local
	    raise OcsbowError('Failed to start manager process: %s' % message)
	ocs.ocsbow.OcsbowError: Failed to start manager process: Could not find host_manager.py in the agent_paths!

To verify my agent paths were sensible, I ran `find $VIRTUAL_ENV -name host_manager.py`, but it outputs nothing, as it appears this file has been renamed, although `ocs/ocsbow.py` still tries to refer to to is explicitly. Manually altering `ocs/ocsbow.py` to refer to `host_manager/agent.py` instead of `host_manager/host_manager.py` seems to have fixed it, allowing the system to start.

BrianJKoopman, 2022-11-10:
Thanks for the bug report, @cnweaver! This should be fixed now in the latest `develop`. It'll be released in ocs v0.10.1.

closed 2022-11-10 by BrianJKoopman

cnweaver, 2022-11-10:
Great, thanks for the fix.

PR #298 (2022-11-10): Update ocs-local-support so it can launch HostManager again

ocs-local-support needs to be able to launch the HostManager, but that bit of code wasn't set up to use the new agent plugin.

## Description

If a user runs "ocs-local-support start agent", the HostManager is launched in the background using python -m ocs.agents.host_manager.agent.  This is used instead of -m ocs-agent-cli, because additional command line args need to be passed through at run time.

## Motivation and Context

Resolves #295

This use case was missed in the recent upgrade to plugin system.  The recommended alternative [... cut]