simonsobs/ocs #300: NaNs / infs in session.data cause trouble in ocs-web

mhasself, 2022-11-08:
NaNs/infs are not compliant JSON, but are passed through from Agents to ocs-web anyway.  This causes ocs-web to totally choke on the session.data ... which is difficult to avoid.

I don't think there's a way to catch this or change how JSON is parsed by autobahn.

Two possible solutions:
1. Convert NaNs/infs to None in ocs_agent.OpSession; this will produce compliant JSON.
2. Reject NaNs/infs entirely -- i.e. forbid them forcibly and raise an exception

I think solution (1) is slightly better, but it should be noted that some Python based clients may be affected by this.  First, one can not distinguish between NaN, inf, -inf.  I think that's ok ... usually we're just trying to flag invalid data, using NaN.

Another consequence of using "None" (which renders as "null" in JSON) is that lists received by Python clients will not as easily be convertible into arrays.  E.g. if the agent sends [1.0, 2.0, None], and the client tries to turn that into a numpy array:
```
>>> np.array([1.0, 2.0, None])
array([1.0, 2.0, None], dtype=object)
```
(that's bad -- it's of dtype=object).

But a work-around on the client side is to specify dtype:
```
>>> np.array([1.,2.,None], dtype=float)
array([ 1.,  2., nan])
```

I think this is an acceptable price to pay, to have ocs-web work and also allow NaNs.

mhasself, 2022-11-08:
Also: I have solution (1) implemented and almost PRed it but thought we should chat about it first ...

https://github.com/simonsobs/ocs/tree/sanitize-session-data

Maybe wants some docs about the np.array business.

BrianJKoopman, 2022-11-10:
I think solution 1 sounds good. I understand the concern with conversion to numpy arrays, though I don't think it's common that Agents store lists in their `session.data` objects, so I'm not sure how often we'll see it. Typical usage is to provide only the latest data point for each field in `session.data`. Though don't let me stop you from documenting the edge case.

The thing I will ask in your current implementation is for you to add some tests for encoding an OpSession that contains Nan, inf, -inf. If it's parameterized and shows working valid types too that's great.

jlashner, 2022-11-11:
If we're already doing some conversion in session.data, do you think we can handle all numpy datatypes? Currently I think whenever you add the result of a numpy computation to session.data you have to convert to int/float, but it would be great if that automatically happened for numpy types.

mhasself, 2022-11-11:
> The thing I will ask in your current implementation is for you to add some tests for encoding an OpSession that contains Nan, inf, -inf. If it's parameterized and shows working valid types too that's great.

I am happy to add tests but honestly I have no idea how/where to put them in the current system... I don't see any OpSession references in the current tests... tips?

mhasself, 2022-11-11:
> If we're already doing some conversion in session.data, do you think we can handle all numpy datatypes? Currently I think whenever you add the result of a numpy computation to session.data you have to convert to int/float, but it would be great if that automatically happened for numpy types.

That sounds doable ... Do you have an example of where this is done?  Do you just mean that you have to .tolist when dropping arrays in session.data?

BrianJKoopman, 2022-11-11:
> I am happy to add tests but honestly I have no idea how/where to put them in the current system... I don't see any OpSession references in the current tests... tips?

I'd put them in `ocs/tests/test_ocs_agent.py`. The structure of the test files tries to mimic the layout of the package with a `test_<filename>` for each file in the package, though deviates in some ways like with the `integration/` directory.

jlashner, 2022-11-11:
Here is an example recently when I had to do it. It's not a huge deal, but just reading / writing code it's very easy to miss that the result of `np.sum(...)` or something similar is not publishable.

https://github.com/simonsobs/socs/blob/c3ba07168be7f4dcf4fc21c84d46d8095eeb6d8f/agents/pysmurf_controller/pysmurf_controller.py#L725-L745

PR #304 (2022-11-14): Sanitize session data

Resolves #300 

## Description

- It's now ok to include numpy arrays / scalars in session.data -- they will be translated to simple lists of python scalars
- It's now ok to include NaN in session.data, it will be converted to None.
- It's now explicitly not ok to include +/- inf in session.data, an internal error will result.
- There are tests for the above.
- The expectations and behavior are documented.

This is a slight "break" in interface in that Inf/-Inf are now expressly forbidden, and NaN will render differently in clients that used to accept it (i.e. Python cli [... cut]
closed 2022-11-14 by BrianJKoopman