simonsobs/ocs #317: `OCSAgent._stop_helper()` warning text specific to "stop"

BrianJKoopman, 2023-02-06:
Initially reported by @laurensaunders.

The warning text within `OCSAgent.stop_helper()` is specific to the "stop" command, and can be confusing if seen when trying to instead register a Task, specifically these lines:
https://github.com/simonsobs/ocs/blob/1c51373471207bb10832ae379f0351b54e0bd01a/ocs/ocs_agent.py#L844-L845

The warning would indicate `agent.register_task()` takes the `stopper_blocking` kwarg, when it really takes `aborter_blocking`.

This, and any similar warnings should be generalized to account for this.

BrianJKoopman, 2023-02-07:
It now seems like what we were running into in debugging adding an aborter to the ACU agent was a missing `@inlineCallbacks` decorator. It seems like the warning here assumes the threading pattern for the aborter is correct, i.e. `@inlineCallbacks` was missing, so this aborter must be blocking, therefore you should register with `stopper_blocking=True`. Where a more useful suggestion would have been to use `stopper_blocking=True`, or consider if the pattern should be non-blocking and is missing `@inlineCallbacks`.

mhasself, 2023-06-26:
@BrianJKoopman Closed via #318?

BrianJKoopman, 2023-06-26:
Yup, closed via #318.

closed 2023-06-26 by BrianJKoopman