simonsobs/ocs #343: HostManager: yaml syntax errors can crash the agent

BrianJKoopman, 2023-08-16:
If a user makes a modification to the SCF or docker-compose file that contains yaml syntax errors, the host manger will crash.

Since the Host Manager runs via systemd, users might not have permissions to restart it after such a crash, so I think we need to make sure this is robust to these sorts of syntax errors, keeping the agent online, but stating the error so that the user can fix and re-run "update" on the agent.

I don't have a good log example of such a crash from when the Agent is running, but I do have an example of if a docker-compose file has a syntax error and the Host Manager is started up:
```
2023-08-16T19:56:35+0000 startup-op: launching manager
2023-08-16T19:56:35+0000 start called for manager
2023-08-16T19:56:35+0000 manager:0 Status is now "starting".
2023-08-16T19:56:35+0000 manager:0 Status is now "running".
2023-08-16T19:56:35+0000 manager:0 CRASH: [Failure instance: Traceback: <class 'yaml.parser.ParserError'>: while parsing a block mapping
  in "/home/ocs/config/lat/daq-lat/docker-compose.yml", line 18, column 3
expected <block end>, but found '<block mapping start>'
  in "/home/ocs/config/lat/daq-lat/docker-compose.yml", line 99, column 4
/home/ocs/.venv/lib/python3.10/site-packages/twisted/internet/defer.py:1697:_inlineCallbacks
/home/ocs/.venv/lib/python3.10/site-packages/ocs/agents/host_manager/agent.py:100:_update_docker_services
/home/ocs/.venv/lib/python3.10/site-packages/twisted/internet/defer.py:1947:unwindGenerator
/home/ocs/.venv/lib/python3.10/site-packages/twisted/internet/defer.py:1857:_cancellableInlineCallbacks
--- <exception caught here> ---
/home/ocs/.venv/lib/python3.10/site-packages/ocs/agents/host_manager/agent.py:453:manager
/home/ocs/.venv/lib/python3.10/site-packages/ocs/agents/host_manager/agent.py:134:_reload_config
/home/ocs/.venv/lib/python3.10/site-packages/ocs/agents/host_manager/agent.py:100:_update_docker_services
/home/ocs/.venv/lib/python3.10/site-packages/twisted/internet/defer.py:1697:_inlineCallbacks
/home/ocs/.venv/lib/python3.10/site-packages/ocs/agents/host_manager/drivers.py:354:parse_docker_state
/home/ocs/.venv/lib/python3.10/site-packages/yaml/__init__.py:125:safe_load
/home/ocs/.venv/lib/python3.10/site-packages/yaml/__init__.py:81:load
/home/ocs/.venv/lib/python3.10/site-packages/yaml/constructor.py:49:get_single_data
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:36:get_single_node
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:55:compose_document
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:84:compose_node
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:133:compose_mapping_node
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:84:compose_node
/home/ocs/.venv/lib/python3.10/site-packages/yaml/composer.py:127:compose_mapping_node
/home/ocs/.venv/lib/python3.10/site-packages/yaml/parser.py:98:check_event
/home/ocs/.venv/lib/python3.10/site-packages/yaml/parser.py:438:parse_block_mapping_key
]
2023-08-16T19:56:35+0000 manager:0 Status is now "done".
^C2023-08-16T19:56:50+0000 Received SIGINT, shutting down.
```

mhasself, 2023-08-16:
I have a backlog of HostManager issues to address, and can include this.  So I'm assigning myself here but anyone should feel free to do a quick fix.

mhasself, 2023-08-18:
I can't get the HostManager to crash due to SCF changes...  the process / update task will crash, but agent seems to stay up.

If the user starts "die", causing HostManager to exit and try to reload ... *that* is vulnerable to an SCF error.  No way to fix that; but we can block users from running that using Access Control (coming soon).

I'll look at the docker-compose next; that file / system is polled much more often than the SCF.

mhasself, 2023-08-18:
I think such fatal errors can only happen when HostManager is _starting_ up.  Once it's running, the agent should stay up even if the manager process crashes.

Still, it is a risk if the config file is invalidated and then someone reboots the computer at some point in the future.

Perhaps we should be fixing this in the systemd definition?

mhasself, 2023-08-18:
For the record, I think this is the answer:
```
 [Unit]
 Description=OCS HostManager for my-host

 [Service]
 ExecStart=/path/to/launcher-hm-grayjay.sh
 User=whatever
 Restart=always
+RestartSec=10s

 [Install]
 WantedBy=multi-user.target
```

BrianJKoopman, 2023-09-01:
> I'll look at the docker-compose next; that file / system is polled much more often than the SCF.

I just was able to reproduce a crash via syntax error in the compose file. Looks the same as the error in the original post (which was also due to a compose file typo.) This results in a crash of the manager process.

mhasself, 2023-09-05:
Ok, but this isn't crashing the _agent_ ... it's crashing the Process.  Once the syntax error is fixed, the process can be started again, from ocs-web or a client.  "Since the Host Manager runs via systemd, users might not have permissions to restart it after such a crash" is not relevant.

However, I will try to incorporate a bit more robustness in the process to this kind of thing.

BrianJKoopman, 2023-09-06:
Yup, good point. I might have also had a SCF error when I logged this issue and concluded I needed an Agent restart. (Assuming that would prevent restarting the Process in a similar manner that it stops the Agent from starting up.)

PR #353 (2023-09-20): HostManager overhaul
## Description

Updates to HostManager to address a number of recent concerns.

Biggest changes:
- New options for instance "manage" setting, such that instances are available for HostManager control, but won't be brought up by default.
- The "manager" process will not crash due to docker-compose.yaml or SCF syntax errors.
- The manager process handles updates to compose files and SCF more gracefully.  If the SCF+compose files are eventually brought into mutual consistency, and "update(reload_config=True)" is run, then the manager process should figure it all out and prese [... cut]
closed 2023-09-20 by BrianJKoopman