simonsobs/ocs #377: Bug in parsing config?

ykyohei, 2024-03-25:
A local discussion with @jlashner led to the conclusion that this might be a bug in ocs.

I found that the parser do not allow the duplicated numbers as follows, during testing hwp-gripper agent https://github.com/simonsobs/socs/pull/646

default.yaml
```
"arguments":
         ["--mcu-ip", "192.168.13.34",
          "--control-port", "8041",
          '--warm-grip-distance', 10.0, 10.0, 15.0,
         ],
```
agent.py
```
pgroup.add_argument('--warm-grip-distance', nargs='+', type=float, default=[10.0, 10.0, 10.0],
                        help='Nominal distance for warm grip position')

Namespace(mcu_ip='192.168.13.34', control_port=8041, warm_grip_distance=[10.0, 15.0], ...
```

Some other test results

- If I put 3 numbers from command line, it accepts duplicated values.

- If I instead do `'--warm-grip-distance', 10.0, 10.00, 15.0`, then I get `Namespace(mcu_ip='192.168.13.34', control_port=8041, warm_grip_distance=[10.0, 10.0, 15.0] ,...`

- If I do `nargs=3` instead of `nargs='+'`, then agent.py raise an error `ocs-agent-cli: error: argument --adjustment-distance: expected 3 arguments`

jlashner, 2024-03-26:
Hi Kyohei,

I tried to reproduce this locally and it's working, as expected... In my site-config I'm using: 
```
      {'agent-class': 'HWPGripperAgent',
       'instance-id': 'hwp-grip',
       "arguments":
         ["--mcu-ip", "192.168.13.34",
          "--control-port", "8041",
          '--warm-grip-distance', 10.0, 10.0, 15.0,
         ]
      },
```
and my args namespace looks like I'd expect it would:
```
Namespace(mcu_ip='192.168.13.34', control_port=8041, warm_grip_distance=[10.0, 10.0, 15.0], adjustment_distance=[-0.5, -0.5, -0.5], supervisor_id=None, no_data_warn_time=60, no_data_shutdown_time=300, site=None, site_file=None, site_host=None, site_hub='ws://localhost:8001/ws', site_http='http://localhost:8001/call', site_realm='test_realm', instance_id='hwp-grip', address_root='observatory', registry_address=None, log_dir=None, working_dir=None, crossbar_timeout=10, agent_class='HWPGripperAgent')
```

Can you give a bit more information about where and how you are running this?

ykyohei, 2024-03-27:
Sorry, the description above was not accurate, 
I thought it generally happens, but it is only happening for adjustment_distance in my env.
```
{
       "agent-class": "HWPGripperAgent",
       "instance-id": "hwp-gripper",
       "arguments":
         ["--mcu-ip", "192.168.13.34",
          "--control-port", "8041",
          '--warm-grip-distance', 10.0, 10.0, 15.0,
          '--adjustment-distance', -0.5, -0.5, -0.2,
         ],
      },
```

```
Namespace(mcu_ip='192.168.13.34', control_port=8041, warm_grip_distance=[10.0, 10.0, 15.0], adjustment_distance=[-0.5, -0.2], supervisor_id=None, no_data_warn_time=60, no_data_shutdown_time=300, site=None, site_file=None, site_host=None, site_hub='ws://localhost:8001/ws', site_http='http://localhost:8001/call', site_realm='test_realm', instance_id='hwp-gripper', address_root='dev', registry_address=None, log_dir='/home/ocs/log/ocs', working_dir=None, crossbar_timeout=10, agent_class='HWPGripperAgent')
```

I made the daq-dev env https://github.com/simonsobs/daq-discussions/discussions/29
and running `ocs-agent-cli --instance-id hwp-gripper`

jlashner, 2024-03-27:
Yep, the minus was confusing the site-config parser. This PR should fix the issue: https://github.com/simonsobs/ocs/pull/378

ykyohei, 2024-03-27:
Great! thank you for your quick fix. I'll close this issue.

closed 2024-03-27 by ykyohei