simonsobs/ocs #380: host-manager failing because no `com.docker.compose.project.working_dir` in docker inspect

jlashner, 2024-04-17:
I'm trying to add a host-manager to control timing and crate monitoring agents. When running the host-manager, I'm getting the following error in the logs:
```
2024-04-17T12:09:59-0700 manager:0 Failed to interpret /home/cryo/repos/ocs-deployment-configs/site/smurf-so9-site/docker-compose.yml and/or its service states: 'com.docker.compose.project.working_dir'
```
Looks like the issue is that these keys don't seem to exist on the smurf-server when I run docker inspect: https://github.com/simonsobs/ocs/blob/90cc862b91c5acb73c435b478c1410685d2e81e9/ocs/agents/host_manager/drivers.py#L425-L426

```
cryo@smurf-so9-site:~/repos/ocs-deployment-configs/site/smurf-so9-site$ docker ps
CONTAINER ID   IMAGE                                   COMMAND                  CREATED      STATUS         PORTS     NAMES
b72864cd785a   simonsobs/socs:v0.4.4-48-gc19e74d-dev   "dumb-init ocs-agent…"   6 days ago   Up 7 minutes             ocs-timingcard
e516f525d5a0   tidair/smurf-tpg-ioc:v5.0.1             "./start_ioc.sh -S s…"   6 days ago   Up 6 days                tpg_ioc
cryo@smurf-so9-site:~/repos/ocs-deployment-configs/site/smurf-so9-site$ docker inspect ocs-timingcard | grep com.docker
                "com.docker.compose.config-hash": "54f3811a035104d897517b2461219745ffc983e7210d7b69ea57ab0368b55af4",
                "com.docker.compose.container-number": "1",
                "com.docker.compose.oneoff": "False",
                "com.docker.compose.project": "smurf-so9-site",
                "com.docker.compose.service": "ocs-timingcard",
                "com.docker.compose.version": "1.23.2"
```

Any ideas why this could be happening?

BrianJKoopman, 2024-04-17:
I think I've seen this before. What version of Docker and Docker Compose are running on that server? I suspect they're old and new versions would include the missing key.

jlashner, 2024-04-17:
Here are the versions:
```
Docker version 24.0.2, build cb74dfc
docker-compose version 1.23.2, build 1110ad01
```
This server is running on ubuntu 22.04, and I believe docker was updated recently (within the last year) but I will try upgrading docker-compose.

BrianJKoopman, 2024-04-17:
Ah, the versions do seem behind. Here's the version info for one of the other site machines running a host manager, also on Ubuntu 22.04:
```
Docker version 26.0.0, build 2ae903e
docker-compose version 1.29.2, build unknown
```

jlashner, 2024-04-17:
Ok, so after upgrading I'm now getting this error: 
```
cryo@smurf-so9-site:~/repos/ocs/ocs/agents/host_manager$ docker-compose ps
Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 33, in <module>
    sys.exit(load_entry_point('docker-compose==1.29.2', 'console_scripts', 'docker-compose')())
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 200, in perform_command
    project = project_from_options('.', options)
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 60, in project_from_options
    return get_project(
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 152, in get_project
    client = get_client(
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 41, in get_client
    client = docker_client(
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 124, in docker_client
    kwargs = kwargs_from_env(environment=environment, ssl_version=tls_version)
TypeError: kwargs_from_env() got an unexpected keyword argument 'ssl_version'
```

This seems to possibly be a breaking change somewhere in docker? Is this something you've seen before?

BrianJKoopman, 2024-04-17:
Hmm, that one I don't think I've seen before. What versions are you on now?

jlashner, 2024-04-17:
Currently this is the version stack on so9:
```
cryo@smurf-so9-site:~/repos/ocs-deployment-configs/site/smurf-so9-site$ docker --version
Docker version 24.0.2, build cb74dfc
cryo@smurf-so9-site:~/repos/ocs-deployment-configs/site/smurf-so9-site$ docker-compose --version
docker-compose version 1.29.2, build unknown
```
It looks like smurf-so3 is currently running a newer version of docker, and docker-compose works there... I may try to upgrade the docker package too:
```
cryo@smurf-so3-satp1:~$ docker --version
Docker version 24.0.7, build afdd53b
cryo@smurf-so3-satp1:~$ docker-compose --version
docker-compose version 1.29.2, build unknown
```

jlashner, 2024-04-17:
Alright, seems like the issue above is because the docker-compose packages we are using are no longer maintained, and have been transferred to "docker": https://docs.docker.com/compose/install/linux/#install-the-plugin-manually

With this update, `docker-compose` has been now replaced by `docker compose`, which breaks lots of our scripts such as host-manager and jackhammer. 

I have a temporary work around by creating an executable `docker-compose` in `~/bin` which I added to the path, now host-manager seems to be working properly when I run it from the command line.

BrianJKoopman, 2024-04-23:
I've opened an issue for adding Compose v2 support to the HostManager in #382. Since this is locally solved by the mentioned work around I'm going to close this issue.

closed 2024-04-23 by BrianJKoopman