simonsobs/ocs #390: Agents running in docker container cannot connect to crossbar when in host network mode

kellyhd, 2024-06-26:
Our network is set up with a wifi card on the host machine which connects to the internet, and a second ethernet network interface (eno1) that connects to a network switch, which then connects to all the devices. From the host, we can communicate with the devices and run the agents locally. We then set up a bridge network (ocs-net) [based on the instructions in the documentation](https://ocs.readthedocs.io/en/main/user/network.html). However, when running the containers on the bridge network, they can only connect to the wifi card network. We confirmed this by pinging google in the container and receiving packets. However, pinging anything on the local network (eno1) does not return any response. When we run our agents we are able to connect to the client, but when we try to initialize a socket connection, it times out because it cannot find the devices connected to eno1.

One thing we found in the [docker networking documentation](https://docs.docker.com/network/) is that 

> A container only sees a network interface with an IP address, a gateway, a routing table, DNS services, and other networking details.

and eno1 does not have all of these (routing table, DNS service, etc) so we think this might be part of why the containers don't recognize it and can't connect to it.

We then tried to run the docker images in host network mode (because we are trying to connect to hardware devices, and the documentation mentions we need to be in host network mode):

> Containers that require communication with networked devices not running Docker (i.e. networked hardware devices such as the Lakeshore 372) will still need to be in the “host” network mode.

 However, when I run the agents (as well as crossbar) on the host network, the agents still can't connect with crossbar, and it's giving me this:

`ControlClientError: [0, 0, 0, 0, 'wamp.error.no_such_procedure', ['no callee registered for procedure <observatory.signalgenerator1>'], {}]`

The logs when I try to run the agent show this:
`2024-06-26T15:55:28+0000 Scheduling retry 1 to connect <twisted.internet.endpoints.TCP4ClientEndpoint object at 0x7f31c53dd970> in 2.0425696452288116 seconds.`

Which seems to indicate that we cannot connect to the crossbar when running in host network mode. We setup the crossbar docker following the instructions in the documentation (with the crossbar started via its own docker-compose file separate from the rest of the agents). I noticed that on the documentation, there was a note that when you put containers in host network mode:

> "Keep in mind that this disables the convenient name resolution provided by the Docker network, so your container will likely require additional configuration, particularly if it needs to also communicate with the crossbar server"

but then it never gives more information about how to configure so the containers can communicate with crossbar. How can I fix this issue?

BrianJKoopman, 2024-07-29:
Hi @kellyhd, apologies for the slow reply, my summer travel schedule has been a bit hectic. I'm not sure I know the answer to your problem, but some thoughts in line below.

> Our network is set up with a wifi card on the host machine which connects to the internet, and a second ethernet network interface (eno1) that connects to a network switch, which then connects to [...]

I do run a configuration somewhat similar to this, with one NIC providing Internet access and another providing a local network over which our OCS system communicates, but I haven't run into this problem before. When you ping something on the local network from within the container, is the target another container running on the network that's connected to the overlay network? If not, it makes sense you can't reach them, since you're isolated in the container. 

> We then tried to run the docker images in host network mode (because we are trying to connect to hardware devices, and the documentation mentions we need to be in host network mode):
>
>
> However, when [...]

When switching between using the bridge network and using "host" mode you might initially have a configuration that specifies the name of the container/service that crossbar is running in, for example (from the [System Configuration example](https://ocs.readthedocs.io/en/main/user/site_config.html#example-config)):
```
    # Address of crossbar within Docker (based on service name)
    'wamp_server': 'ws://crossbar:8001/ws',
    'wamp_http': 'http://crossbar:8001/call',
```

When you move to "host" mode this name won't resolve to point to the crossbar server anymore. You have a few options, but perhaps the most direct is to use the IP (on the "host" network) of the machine running your crossbar server, for example if your host machine (sounds like on `eno1`) is 192.168.1.2:
```
    'wamp_server': 'ws://192.168.1.2:8001/ws',
    'wamp_http': 'http://192.168.1.2:8001/call',
```
Setting the crossbar address within the context of your network configuration is all that is meant by "additional configuration" here. Perhaps that could be clarified in the docs.

Since it's been a month, I'm wondering if you've managed to get things working, or is this still an issue? If you've got things working, a quick update here might be helpful to others. If not, it's worth considering the address resolution mentioned above. You might also check out the [docker network create](https://docs.docker.com/reference/cli/docker/network/create/) documentation for when creating your overlay network in the event you need to specify additional options based on your network configuration, like the gateway.

kellyhd, 2024-07-29:
Yeah we fixed the issue! The solution was that on each device connected to eno1, we had configured the IP address and subnet mask, but we also had to configure the default gateway to be eno1's IP address. I guess

closed 2024-07-29 by kellyhd

kellyhd, 2024-07-29:
Yeah we fixed the issue! I think we had configured the IP address and subnet mask on each device, but we did not set the default gateway to eno1's IP address, and once we did that the containers could run. I guess the devices could only communicate locally with the same network, allowing the agent to run locally, but couldn't reach the containers which were on a separate network. We didn't end up needing to put the containers in host network mode. The documentation did say that agents for networked hardware devices like Lakeshore 372 need to be in host network mode but maybe our system is different.